NETMF/netmf-interpreter #502: STM32F429IDISCOVERY Reports no SPI

piwi1263, 2016-08-24:
When trying to get the LCD to work via SPI I ran into a fail. After checking the SPI port count I got zero back as can be seen in the attached window. The hardware provider has 4 SPI Modules in its enumeration. I've got no clou where these come. Any ideas / tips on the SPI side of things ?

[image]

cw2, 2016-08-25:
The solution uses SPI stubs and thus the hardware provider returns zero SPI ports. You'd need to replace the following lines in TinyCLR.proj
```
<DriverLibs Include="cpu_spi_stubs.$(LIB_EXT)" />
<RequiredProjects Include="$(SPOCLIENT)\DeviceCode\Drivers\Stubs\Processor\stubs_SPI\dotNetMF.proj" />
```
with
```
<DriverLibs Include="STM32F4_SPI.$(LIB_EXT)" />
<RequiredProjects Include="$(SPOCLIENT)\DeviceCode\Targets\Native\STM32F4\DeviceCode\STM32F4_spi\dotNetMF.proj" />
```
Then hardware provider should return correct number of SPI ports, based on the configuration provided in platform_selector.h
```
// SPI (LCD-SPI and L3GD20 gyroscope on SPI5)
//                                         SPI1      SPI2     SPI3      SPI4      SPI5      SPI6
#define STM32F4_SPI_SCLK_PINS           { _P(A, 5), _P_NONE_,_P_NONE_, _P_NONE_, _P(F, 7), _P_NONE_ }
#define STM32F4_SPI_MISO_PINS           { _P(A, 6), _P_NONE_,_P_NONE_, _P_NONE_, _P(F, 8), _P_NONE_ }
#define STM32F4_SPI_MOSI_PINS           { _P(A, 7), _P_NONE_,_P_NONE_, _P_NONE_, _P(F, 9), _P_NONE_ }
```
I guess SPI6 should be removed...

_Note: The current implementation of most NETMF peripherals does not allow disjointed ranges, i.e. it is not possible to configure only SPI1 and SPI5, without the others between them._

piwi1263, 2016-08-26:
Well, that was relative easy. 

Made the changes as said above. 

Looking at <root>\DeviceCode\Targets\Native\STM32F4\ManagedCode\Hardware\CPU.cs there is only mention of 3 SPI_Devices. SDK 4.4 has 4 so I had the trick the 5th SPI:
[image]
and by using the following in the platform_selector.h:
[image]
After a rebuild of the TinyClr, flashing and running the software again, the LCD was back to live:
[image]

@CW2: Thank you for the help.

smaillet-ms, 2016-08-26:
W00t! W00t!

closed 2016-08-30 by piwi1263