NETMF/netmf-interpreter #507: Micro Framework project general status

tdjastrzebski, 2016-09-04:
I am sorry for posting here but could not find any active forum.
Is NETMF project active? The last blog post was a year ago, I am unable to find any microcontroller supporting version 4.4. Only 4.3 version and I am not willing to install required VS 2013 again.
Where should I look, ask? I would like to try I out. Please advise.

piwi1263, 2016-09-04:
Of course it depends on your expectations considering the needed resources and ports avail, but I've got good experiences with the F4 series from ST. Mainly the STM32F411, but you'll find a lot of efforts for other MCU in the ST family, even there is the start of porting NetMf to the STM32F7xx if you read through the gitter room https://gitter.im/NETMF/netmf-interpreter?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge&utm_content=badge you'll find all kinds of efforts undertaken. Or follow this #502 #482

tdjastrzebski, 2016-09-04:
I have not been playing with microcontrollers for like 20 yrs but what I need is probably STM32F412 since my project will require CAN Bus. Is NETMF 4.4 supported on this chip?
In any case, it is sad to see MS abandoned active involvement but that is just the way it goes.

piwi1263, 2016-09-04:
I do have some experimental, partly port to the 412, so yes I does run 4.4. At the moment I only have SPI checked and displaying some information on an ILI9431 based LCD using VS2015 and 4.4. No CAN checked yet.
[image]