NETMF/netmf-interpreter #511: Remove obsolete code for native patching

cw2, 2016-09-06:
Unfortunately, I cannot find the original source (it could be either a CodePlex issue or a topic on the old netmf.com discussion forum), but IIRC the mechanism for native patching is obsolete since 4.x (and maybe even before that).

I'd suggest removing this code, namely handling of `patch_*` parameters in `WatchAssemblyBuilder::Linker::Generate(...)` function and related command line arguments parsing in MetadataProcessor.cpp.

Additional benefit will be removal of three cryptolib function calls (`Crypto_GetHash`,`Crypto_PublicKeyFromModulus` and `Crypto_RSADecrypt`), significant reduction of crypto functions needed on the device side (they are then called only from stubs and/or _test_win_ test application).

smaillet-ms, 2016-09-06:
Yes, patching has never been officially documented for the public releases of NETMF. It was something that was part of the original product that NETMF was used to create but not carried forward. As mentioned in the [docs in the v5_dev branch](https://github.com/NETMF/netmf-interpreter/blob/v5_dev/Docs/BookOfTheRuntime/AssemblyHeader.md#flags) The patch option is considered obsolete and not to be used. In vNext all support for that should be stripped out.

cw2, 2016-09-06:
How about do it now?

smaillet-ms, 2016-09-06:
At this point it does no harm to leave it in and doesn't really add value to current users to remove.  My plan for MetadataProcessor is to actually swap out the version we have now for the C# version from the [Xamarin/Monkey robotics](https://github.com/xamarin/Monkey.Robotics/tree/master/Source/Xamarin%20Studio%20Microframework%20Add-in/MFMetaDataProcessor) version. As it will be easier to handle the additional functionality desired when it's all C#. This also means that MetadataProcessor won't be one big giant "Swiss Army Knife" tool anymore either. There, were a number of things like the signature generation and other crypto support in MetadataProcessor that had nothing to do with actually processing the metadata.

Given that - making changes to MetadataProcessor that aren't causing actual real problems doesn't make sense to me.

cw2, 2016-09-06:
Ok, thanks for the feedback.

closed 2016-09-06 by cw2