NETMF/netmf-interpreter #514: Change CPU_GPIO_DisablePin AltFunction parameter type to UINT32

cw2, 2016-09-14:
As I have hit similar issue with the GPIO API, I vote for @smaillet-ms suggestion in #464 ([comment](https://github.com/NETMF/netmf-interpreter/issues/464#issuecomment-223826334)) to change the type of the alternate parameter in `CPU_GPIO_DisablePin` function from `GPIO_ALT_MODE` to `UINT32`, and add inline function for backward compatibility:

```C++
void CPU_GPIO_DisablePin(GPIO_PIN Pin, GPIO_RESISTOR ResistorState, UINT32 Direction, UINT32 AltFunction);

inline void CPU_GPIO_DisablePin(GPIO_PIN Pin, GPIO_RESISTOR ResistorState, UINT32 Direction, GPIO_ALT_MODE AltFunction)
{
  return CPU_GPIO_DisablePin(Pin, ResistorState, Direction, (UINT32)AltFunction);
}
```

cw2, 2016-09-14:
Alternatively, change the enum underlaying type
```C++
enum GPIO_ALT_MODE : UINT32
{
...
```

smaillet-ms, 2016-09-14:
Well, that's even simpler than the idea of using a strongly typed enum - works for me! Send a PR, but we'll need to get the build succeeding on all platforms before it can be merged in. (At this point in time The GCC builds for MCBSTM32F400 are failing).

cw2, 2016-09-14:
Does that work in Keil? I am not sure which version supports [C++11] strongly typed enums.

Specifying the underlying type (enum A : xxx) results in Error:  #70: incomplete type is not allowed.

smaillet-ms, 2016-09-14:
It requires some compiler settings so is more work to get functional in NETMF. The solution posted here skips all that rather nicely so I wouldn't bother with the strongly typed enum.

cw2, 2016-09-14:
Oh, I misread your previous comment, sorry. I'll have a look at what's wrong with GCC for MCBSTM solution (so I don't have broken build in PR : )...