NETMF/netmf-interpreter #520: [ANN] Debug .NETMF native code using VS Code

josesimoes, 2016-10-20:
This is a guide to use VS Code as a debugging environment for NETMF native code.
Not a competition or wanting to pick on @cw2 ;) !
The motivation behind it is that I’ve been using VS Code for native code development for quite some time now, and felt that being able to debug it right there would be a nice improvement and would boost productivity.
Anyways, it’s another option for the community members interested in this side of the development.

[image]

Requirements:
-	VS Code installed
-	C/C++ VS Code extension 
-	OpenOCD installed (pretty much any working distribution will work)
-	ST-LINK (either the device or the equivalent circuit in the development board)

Add the following file (launch.json) in the .vscode directory at the root folder of the NETMF repository .

```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": ".NETMF core debug (Windows)",
            "type": "cppdbg",
            "request": "launch",
            "miDebuggerPath": "E:/GNU_Tools_ARM_Embedded/5_4_2016q2/bin/arm-none-eabi-gdb.exe",
            "targetArchitecture": "ARM",
            "program": "E:/GitHub/netmf-interpreter/BuildOutput/THUMB2FP/GCC5.4/le/FLASH/release/STM32F4DISCOVERY/bin/tinyclr.axf",
            "setupCommands": [
                {"text": "target extended-remote localhost:3333"},
				{"text": "file E:/GitHub/netmf-interpreter/BuildOutput/THUMB2FP/GCC5.4/le/FLASH/release/STM32F4DISCOVERY/bin/tinyclr.axf"},
				{"text": "monitor reset halt"}
            ],
            "customLaunchSetupCommands": [
			    {"text": "monitor reset init"}
            ],
            "launchCompleteCommand": "exec-continue",
			"debugServerPath": "C:/Program Files (x86)/openocd-0.10.0/bin/openocd.exe",
			"debugServerArgs": "-s \"C:/Program Files (x86)/openocd-0.10.0/bin/scripts/\" -f interface/stlink-v2-1.cfg -f board/stm32f4discovery.cfg",
			"serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "filterStderr": true,
            "externalConsole": true,
            "cwd": "${cwd}"
        }
    ]
}
```

A few explanations on what you need to tweak there.
miDebuggerPath: full path to the gdb executable.
program: full path to the .axf output file that results from a successful NETMF build.
setupCommands (the second ‘text’ entry): the same as the previous one.
debugServerPath: full path to the OpenOCD executable.
debugServerArgs: full path to the scripts directory on the OpenOCD installation AND the appropriate .cfg files for the interface and the board. In this example I’m using the STM32F4DISCOVERY board, so I add there the ‘stlink-v2-1.cfg’ for the interface and ‘stm32f4discovery.cfg’ for the board support configuration.
(note the forward slash on all the paths there)

And that’s it! 
Assuming that you have successfully build a NETMF image, have loaded it on the board and that you’ve open the root folder of the NETMF repository with VS Code, you can navigate to the View – Debug menu. You’ll see there the debug interface and the familiar green play button (or F5, if you prefer). Hit that one and the magic will happen.
You can set breakpoints, add watch variables, navigate up and down the call stack and use the toolbar or the familiar keyboard shortcuts to debug the code.
Final note: to have enough info in the compiled output for a decent debug session make sure you set the appropriate flavour to the build command. As an alternative you can always tweak the Microsoft.Spot.system.gcc.targets file. I’ve changed mine in these two lines:

`<CC_CPP_TARGETTYPE_FLAGS Condition="'$(FLAVOR)'=='Release'"      >$(CC_CPP_TARGETTYPE_FLAGS) -Os -g3 -ggdb</CC_CPP_TARGETTYPE_FLAGS></code>`

`<ARCH_TARGETTYPE_FLAGS Condition="'$(FLAVOR)'=='Release'"      >-Os -g3 -ggdb</ARCH_TARGETTYPE_FLAGS></code>`

An improvement that is on the way is to be able to launch the build and load the ouput image. All this without leaving VS Code.

piwi1263, 2016-10-20:
Nice work, I'll definately try this one. Launch time of the full VS is so .....

maiorfi, 2016-10-21:
👍

techcap, 2016-10-23:
Very nice!!! Do you know how to integrate with compile key(Ctrl+Shift+B)?

josesimoes, 2016-10-24:
@techcap I know the way there, just hadn't have the time to complete. As I've mention at the end of the initial post, I'm working to have the complete build/deploy/debug in VS Code.

josesimoes, 2016-10-26:
@techcap yes I do now! 😛

Glober777, 2017-03-12:
@josesimoes, I'm a bit new to NETMF, and I'm wondering if the following could also be used to manage Netduino board, any idea?

techcap, 2017-03-17:
@josesimoes I was suffered because of debugger disconnection after starting debug mode. I found it was because of power mode which you included in your PR #524.
I really appriciate for your help. Thank you very very much.