NETMF/netmf-interpreter #528: PWM on STM32F4DISCOVERY

vgolovanov, 2016-11-21:
Hi.
I'm facing a problem with PWM code on STM32F4DISCOVERY. 
I tried different variations for pin assignment, but problem the same. 
At the moment of PWM object initialization, i'm getting error: "An unhandled exception of type 'System.NotSupportedException' occurred in Microsoft.SPOT.Hardware.PWM.dll".

Also i tried to step in by debugger and get more detailed error, but that information doesn't help me a lot.

```
Found debugger!

Create TS.

 Loading start at 80502ec, end 80646c8

   Assembly: mscorlib (4.4.0.0)     Assembly: Microsoft.SPOT.Native (4.4.0.0)     Assembly: Microsoft.SPOT.Hardware (4.4.0.0)  
   Assembly: Microsoft.SPOT.Hardware.Usb (4.4.0.0)     Assembly: Microsoft.SPOT.Hardware.SerialPort (4.4.0.0)  
   Assembly: Windows.Devices (4.4.0.0)  Loading Deployment Assemblies.

Attaching deployed file.

   Assembly: System.IO (4.4.0.0)  Attaching deployed file.

   Assembly: PwmTesting (1.0.0.0)  Attaching deployed file.

   Assembly: Microsoft.SPOT.IO (4.4.0.0)  Attaching deployed file.

   Assembly: Microsoft.SPOT.Hardware.PWM (4.4.0.0)  Resolving.

The debugging target runtime is loading the application assemblies and starting execution.
Ready.

'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.Native.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.Hardware.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.Hardware.Usb.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.Hardware.SerialPort.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Windows.Devices.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.IO.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\System.IO.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.4\Assemblies\le\Microsoft.SPOT.Hardware.PWM.dll', Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'Microsoft.SPOT.Debugger.CorDebug.14.dll' (Managed): Loaded 'C:\projects_tfs\PwmTestingNetMF\PwmTesting\bin\Debug\le\PwmTesting.exe', Symbols loaded.
    #### Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (2) ####
    #### Message: 
    #### Microsoft.SPOT.IO.RemovableMedia::.cctor [IP: 001d] ####
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.HwProvider.get'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.HardwareProvider'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.HardwareProvider'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.HwProvider.get'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.GetPwmPinForChannel'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.HardwareProvider.GetPwmPinForChannel'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PeriodFromFrequency'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.DurationFromDutyCycleAndPeriod'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
    #### Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (1) ####
    #### Message: 
    #### Microsoft.SPOT.Hardware.PWM::.ctor [IP: 0041] ####
    #### PwmTesting.Program::Main [IP: 0018] ####
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.PWM'
Step into: Stepping over non-user code 'Microsoft.SPOT.Hardware.PWM.Dispose'
    #### Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (1) ####
    #### Message: 
    #### Microsoft.SPOT.Hardware.PWM::Dispose [IP: 0004] ####
    #### PwmTesting.Program::Main [IP: 0018] ####
    #### Exception System.NotSupportedException - CLR_E_NOT_SUPPORTED (1) ####
    #### Message: 
    #### Microsoft.SPOT.Hardware.PWM::Dispose [IP: 000f] ####
    #### PwmTesting.Program::Main [IP: 0018] ####
Exception thrown: 'System.NotSupportedException' in Microsoft.SPOT.Hardware.PWM.dll
An unhandled exception of type 'System.NotSupportedException' occurred in Microsoft.SPOT.Hardware.PWM.dll

The program '[1] Micro Framework application: Managed' has exited with code 0 (0x0).
```

Any help or just direction where to dig, is much appreciated.
Best regards,
Vyacheslav.

IngenuityMicro, 2016-11-22:
The current STM solution only has 4 PWM channels (PD12-PD15) assigned.

Are you using something like:

`private static PWM _pwm = new PWM(Cpu.PWMChannel.PWM_0, 4000, 0.5, false);`

vgolovanov, 2016-11-22:
Yes, i used the code like this with different variations Cpu.PWMChannel.PWM_0, with channel numbers from PWM_0 to PWM_10.
Also i tried to change pins assignment in platform file. I tried already compiled firmware from other source. Also yesterday i compiled firmware for Disco429 and result the same.

IngenuityMicro, 2016-11-22:
Add this to TinyCLR.proj

`<Import Condition="'$(PWM_FEATUREPROJ)'==''" Project="$(SPOCLIENT)\Framework\Features\PWM.featureproj" />`

vgolovanov, 2016-11-22:
Hi Justin.

That solved my problem. :-)
Thank you very much for your time and help.

Best regards,
Vyacheslav.