NETMF/netmf-interpreter #537: Is .Net MF being discontinued?

Korporal, 2017-02-07:
I'm getting mixed messages as I browse the web and do not have any solid idea if MS intend to continue investing in .Net Micro Framework or if it is deemed an experiment that has no long term future, can anyone involved comment?

ianlee74, 2017-02-07:
Microsoft hasn't made any investments in NETMF in several years and they've indicated that they have zero employees working on the project and have no plans to add any in the future. There are a few Microsoft employees that work on it from time to time on their own time. So, this is fully a community run project now with the exception that they still own this repo. Fork away! 

If you're looking for a supported version of NETMF, I'd check out GHI's new TinyCLR products. Unfortunately, it is not open source.

aalmada, 2017-03-09:
Is there any plan/intention to join the .NET Foundation?

smaillet-ms, 2017-03-10:
NETMF is already a apart of .NET Foundation. Due to historical reasons it isn't under the GitHub dotnet organization. (It is possible to change that, though it isn't without potential disruptions as all local repos would need to reset their upstream "remote". Thus far there, hasn't been a significant driver for doing that. But if a rationale argument can be made and enough regular users agree it's not out of the question to shift that.