NETMF/netmf-interpreter #538: Visual Studio 2017

matheusssilva, 2017-03-09:
Is there a forecast to add support for visual studio 2017?

Glober777, 2017-03-09:
I'm interested in that as well

techcap, 2017-03-15:
👍

0x0001F36D, 2017-05-22:
+1 , I bought the Netduino3-wifi few days ago , but idk how to use it in my visual studio 2017 , Is anyone have same problem ?

networkfusion, 2017-05-22:
Use VS2015 or follow progress at GitHub.com/nanoframework not much has been going on here for a long time...

miloush, 2017-05-31:
Alright, there is still a thing or two I wanted to sort out but you can use https://github.com/miloush/netmf-interpreter/tree/VS15.3 which builds on and for Visual Studio 2017. Building requires 15.3 (26510.0-Preview works) and unlike previously, you need to run the build script from the developer command prompt (ensure the respective instance has _Desktop development with C++_, _.NET desktop development_ and _Visual Studio extension development_ workloads installed). This is a 4.4 version based on `dev` branch.

Tridy, 2017-06-06:
Have the VS2017 - 15.3 (26510.0-Preview) installed with the components:

- Desktop development with C++
- .NET desktop development
- Visual Studio extension development

got the latest dev source:
Latest commit 03d84fe  on Sep 6, 2016

extracted netmf-interpreter-VS15.3 folder into c:\Build
ran powershell scripts:

- c:\Build\netmf-interpreter-VS15.3\Install-BinTools.ps1
- c:\Build\netmf-interpreter-VS15.3\Install-CMSIS.ps1
- c:\Build\netmf-interpreter-VS15.3\Update-DeviceSdkAssembly.ps1

then i run build_sdk.cmd but get errors:
[build.zip](https://github.com/NETMF/netmf-interpreter/files/1054705/build.zip)

what am I missing?

thanks

miloush, 2017-06-06:
Crypto libraries! https://github.com/NETMF/netmf-interpreter/releases/download/NETMFCryptoLibraries/NetMFCryptoLibs.msi

Tridy, 2017-06-06:
Thanks! It worked.

- downloaded 6a3d134,
- downloaded the _NetMFCryptoLibs.msi_ installed it and copied the _lib_ folder to the _crypto_ folder
- then the build went through

After that, installed the vsix from \BuildOutput\Public\Release\Server\dll\NetmfVS15.vsix 
As a result, it is possible to create a new _Micro Framework_ project.

[image]

Next step to see how to make it work with Netduino 3.

miloush, 2017-06-06:
Note that if you haven't installed the MF SDK, you need to go to BuildOutput\Public\Release\Server\msm and install the msi, otherwise the projects won't compile. :-)

Tridy, 2017-06-06:
I see.

So, after a successful build, there are 2 files to install for VS to work:

- \BuildOutput\Public\Release\Server\msm\MicroFrameworkSDK.MSI (Micro Framework itself)
- \BuildOutput\Public\Release\Server\dll\NetmfVS15.vsix (VS Extension)

I think the official 4.4 version from Oct 20, 2015 that I had already installed [MicroFrameworkSDK.MSI](https://github.com/NETMF/netmf-interpreter/releases/download/v4.4-RTW-20-Oct-2015/MicroFrameworkSDK.MSI) worked as well, but I uninstalled it and installed the one from my local build instead, just to make sure.

Thanks!

Tridy, 2017-06-06:
I got the framework, SDK and Templates working in VS2017. The problem now is that when I try to deploy to my _Netduino 3 WiFi_, it says:

> Severity	Code	Description	Project	File	Line	Suppression State
Error		Cannot deploy the base assembly 'mscorlib', or any of his satellite assemblies, to device - USB:Netduino twice. Assembly 'mscorlib' on the device has version 4.3.1.0, while the program is trying to deploy version 4.4.0.0				

which, as far as I understand, means that the firmware on the board has framework 4.3.1.0 and I am trying to deploy for 4.4.0.0

Could someone help m with compiling the new firmware baser on framework 4.4 for my Netduino 3 WiFi? Any hints?

Thanks!

techcap, 2017-06-07:
I think you installed SDK v4.3.1. Uninstall it and install SDK v4.4

Tridy, 2017-06-07:
Netduino firmware is framework specific. So, the application framework version must match the firmware version. I cannot deploy a framework to the device directly. I need to recompile and deploy the firmware with the matching one. Otherwise I get this:  

[image]

so, I need some help with recompiling the firmware

thanks!

miloush, 2017-06-07:
It might be easier to get the MF 4.3 up and running in VS. Have you tried to install the 4.3 MF SDK? Fingers crossed it might give you the ability to switch to 4.3 target framework in the project properties while still using the existing 15.3 VS template.

Tridy, 2017-06-07:
good idea, but that did not go that well.
i needed to tweak the proj file a bit but in the end, targeting 4.3 would not allow having any references:

[image]

it seems like Netduino site has been on hold for the last year or so, including the community. It might be so that [wildernesslabs](https://community.wildernesslabs.co/) could pick it up some time soon.

miloush, 2017-06-07:
OK so I am not sure what you have tried, but what the 4.3 SDK installs into C:\Program Files (x86)\MSBuild\.NET Micro Framework\v4.3 must go into C:\Program Files (x86)\Microsoft Visual Studio\[your VS 2017 instance]\Community\MSBuild\Microsoft\.NET Micro Framework. Do you have that? Otherwise the libraries should be working, they are referenced all at HKLM\Software\WOW6432Node\Microsoft\.NETMicroFramework

Tridy, 2017-06-07:
I see.
The registry settings [registry_settings.zip](https://github.com/NETMF/netmf-interpreter/files/1059381/registry_settings.zip) for both 4.3 and 4.4 are poining to the libraries in
 C:\\Program Files (x86)\\Microsoft .NET Micro Framework\\

I also copied the v4.3 from that folder into:

c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\MSBuild\Microsoft\.NET Micro Framework\v4.3\

the thing now is that I can create a 4.4 project:

[image]

but I cannot change it to 4.3:

[image]

I have both 4.3 and 4.4 installed:

[image]

but when trying to use 4.3 in VS 2017 it says

> The reference assemblies for framework ".NETFramework,Version=v4.3" were not found. To resolve this, install the SDK or Targeting Pack for this framework version or retarget your application to a version of the framework for which you have the SDK or Targeting Pack installed. Note that assemblies will be resolved from the Global Assembly Cache (GAC) and will be used in place of reference assemblies. Therefore your assembly may not be correctly targeted for the framework you intend.	MFConsoleApplication1			

Somehow VS2017 cannot see it.

I tried to reinstall 4.3 SDK but get the same problem.

What am I missing?

Thanks!

miloush, 2017-06-07:
Not sure, I will have to try this. You said you had to tweak the project a bit, what did you do?

doingnz, 2017-06-08:
To use 4.4 you need 4.4 firmware inside the device. Do you have access to the custom firmware source for the HW Netduino 3 WiFi to rebuild it? 

I did not see anything suitable here: http://www.netduino.com/downloads/
.
If you don't have the specific solution files needed for the HW Netduino 3 WiFi, then you need to port one of the existing solutions to that HW. (CPU and peripheral drivers needed as per the HW) 

Otherwise you need to use NetMF 4.3 and the SDKs as available here: http://www.netduino.com/downloads/

hope this helps.

Tridy, 2017-06-08:
in the project, replace the imports:

`  <Import Project="$(NetMfTargetsBaseDir)$(TargetFrameworkVersion)\CSharp.Targets" />
`
with
`
  <Import Condition="EXISTS('$(NetMfTargetsBaseDir)$(TargetFrameworkVersion)\CSharp.Targets')" Project="$(NetMfTargetsBaseDir)$(TargetFrameworkVersion)\CSharp.Targets" />
<Import Condition="!EXISTS('$(NetMfTargetsBaseDir)$(TargetFrameworkVersion)\CSharp.Targets')" Project="$(MSBuildToolsPath)\Microsoft.CSharp.targets" />
`

I found this difference with the new 4.4 project, so I copied that into 4.3 project and it seems to open but the rest I described in the previous posting.

It seems indeed, that I should make VS 2017 work with 4.3 instead of going for 4.4, since that is the on the device.

doingnz, 2017-06-08:
maybe ask if anyone has progressed the code as discussed here: 

http://forums.netduino.com/index.php?/topic/11796-from-build-2015-netduino-3-net-mf-44-universal-windows-platform-api-convergence-and-alljoyn/

miloush, 2017-06-08:
Hmm it seems to work for me. I created 4.4 project as you did, got the same error, copied 4.3 targets to the VS folder, tried to add some references and it has built and run (in the emulator).

Your VSIX installer shows 2 instances of VS, are you sure you copied the 4.3 targets to the right instance?

Tridy, 2017-06-08:
I have both the RTM and Preview instances of 2017 installed.
I will remove the RTM and stay just with the Preview and start over with the process,

miloush, 2017-06-08:
not sure that is necessary, you can copy the targets to both :)

Tridy, 2017-06-08:
I uninstalled both in the end and installed just the preview, including the 3 components mentioned before.

it was actually a [newer preview](https://www.visualstudio.com/en-us/news/releasenotes/vs2017-preview-relnotes) (good timing, right?) with version 26606.0 that was installed. The installation folder changed as well

from: c:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\
to:     c:\Program Files (x86)\Microsoft Visual Studio\Preview\Enterprise\

i cleaned up the folders, copied the source again and tried to build. This time I am getting some red text, including:
C:\Build\NETMF\netmf\Framework\CorDebug\VsPackage.cs(174,22): error CS1748: Cannot find the interop type that matches the embedded interop type 'Microsoft.VisualStudio.Shell.Interop.IVsToolboxItemProvider2'. Are you missing an assembly reference? [C:\Build\NETMF\netmf\Framework\CorDebug\vs15\CorDebugvs15.csproj]

miloush, 2017-06-08:
I have not tried building on this build but have you checked the VS SDK workload?

Tridy, 2017-06-08:
Here is a full log of the build: [build2.zip](https://github.com/NETMF/netmf-interpreter/files/1062457/build2.zip).

_VS SDK Workload_, is it [this](https://docs.microsoft.com/en-us/visualstudio/install/workload-component-id-vs-community) page you are talking about? I do not think it was updated for the latest build. Could I ask what I do with the _version_ and _dependency_ type info that I find there? 

Thanks.

miloush, 2017-06-08:
I just meant whether you did not forgot to install it when you reinstalled the VS (i.e. you checked managed development, native development and VS SDK in the installer), but the log suggests the problem is a bit different. Could you share the sdkpre.log?

Tridy, 2017-06-09:
Here it comes: 
[sdkpre.zip](https://github.com/NETMF/netmf-interpreter/files/1063059/sdkpre.zip)

miloush, 2017-06-09:
I confirm I reproduced the error in the recent VS build, that's the risk when targeting pre-releases! Investigating.

miloush, 2017-06-09:
Yep, there were changes. Try this one :) https://github.com/NETMF/netmf-interpreter/releases

You will have to delete the BuildOutput directory and build again.

Tridy, 2017-06-11:
tried it, got some errors. 
[log3.zip](https://github.com/NETMF/netmf-interpreter/files/1066327/log3.zip)
have I missed something?

miloush, 2017-06-11:
Well that build looks very broken... looks like you are missing the tools in the parent directory?

Tridy, 2017-06-11:
I do have the tools in:
C:\Build\NETMF\tools\x86
and the cmd is: 
c:\Build\NETMF\netmf\build_sdk.cmd

I tried to run it a couple more times and I get the same error.

when I look at the first error part of the log, I see:

```
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error VSSDK1028: Can
not find the tools for VS SDK. [C:\Build\NETMF\netmf\Framework\CorDebug\vs14\CorDebugvs14.csproj]
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error MSB4018: The "C
reatePkgDef" task failed unexpectedly.\r [C:\Build\NETMF\netmf\Framework\CorDebug\vs14\CorDebugvs14.csproj]
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error MSB4018: System
.InvalidOperationException: Cannot start process because a file name has not been provided.\r [C:\Build\NETMF\netmf\Fra
mework\CorDebug\vs14\CorDebugvs14.csproj]
```

Could it be caused by CorDebug\vs**14**\ and not CorDebug\vs**15**\ being used there?

miloush, 2017-06-11:
The build process tries to build SDK with every VS installed. It looks like you have Visual Studio 2015 installed but not the SDK. However, that shouldn't matter to you if you are interested in 2017 output only.

The first error I have seen in your logfile above is
```
C:\Program Files (x86)\Microsoft Visual Studio\Preview\Enterprise\MSBuild\15.0\bin\Microsoft.Common.CurrentVersion.targets(1978,5): warning MSB3245: Could not resolve this reference. Could not locate the assembly "CODESIGN.Submitter". Check to make sure the assembly exists on disk. If this reference is required by your code, you may get compilation errors. [C:\Build\NETMF\netmf\Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj]
BuildSigner.cs(7,7): error CS0246: The type or namespace name 'CODESIGN' could not be found (are you missing a using directive or an assembly reference?) [C:\Build\NETMF\netmf\Framework\Tools\BuildTasksInternal\BuildSigner\BuildSignerSpotBuild.csproj]
```

Which should be pointing to C:\Build\NETMF\tools\x86\CODESIGN\CODESIGN.Submitter.dll. If you have the file there, it means the $(SPOROOT) variable is not pointing at C:\Build\NETMF.

When you run the build_sdk and then list the variables with `set` command, do you have `SPOROOT`  there?

I am a bit puzzled that you have built it before without problems.

Tridy, 2017-06-11:
[image]

SPOROOT=C:\Build\NETMF

[set_variables.txt](https://github.com/NETMF/netmf-interpreter/files/1066715/set_variables.txt)

note that I stopped the execution after the first red line with errors appeared (just in case if some of the variables are reassigned to different values in the process)

miloush, 2017-06-11:
Oh well, that's strange indeed. I guess the sdkpre log needs to be checked why it can't find it then. (Also I don't get any of the nuget warnings...)

Tridy, 2017-06-11:
Here it comes.

    248 Warning(s)
    16 Error(s)

Time Elapsed 00:09:24.05

[sdkpre.zip](https://github.com/NETMF/netmf-interpreter/files/1066723/sdkpre.zip)

miloush, 2017-06-11:
Oh so you don't have the CODESIGN problem anymore. Before starting the build, can you `set VSSDK14INSTALLDIR=` and see how it goes?

Tridy, 2017-06-11:
I tried it, still got the error. 

```
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error VSSDK1028: Can
not find the tools for VS SDK. [C:\Build\NETMF\netmf\Framework\CorDebug\vs14\CorDebugvs14.csproj]
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error MSB4018: The "C
reatePkgDef" task failed unexpectedly.\r [C:\Build\NETMF\netmf\Framework\CorDebug\vs14\CorDebugvs14.csproj]
C:\Program Files (x86)\MSBuild\Microsoft\VisualStudio\v15.0\VSSDK\Microsoft.VsSDK.targets(961,5): error MSB4018: System
.InvalidOperationException: Cannot start process because a file name has not been provided.\r [C:\Build\NETMF\netmf\Fra
mework\CorDebug\vs14\CorDebugvs14.csproj]
```

[sdkpre.zip](https://github.com/NETMF/netmf-interpreter/files/1066775/sdkpre.zip)

I don't really get why I get this VS SDK problem.

I have 2015 installed to see if everything is actually working as expected in 2015. And it does.

miloush, 2017-06-11:
I don't get it either. What I don't get is that it tries to build the CorDebugvs14 which it only should do if there is VS 2015 SDK, see Support\FrameworkBase\dotnetmf.proj. but then it doesn't seem to find it. I will probably have to setup a machine with both VS 2015 and VS 2017 and try to repro. Meanwhile you could leave only the CorDebugvs15.csproj reference in that dotnetmf.proj if it helps. Of course, if you get the same errors for CorDebugvs15, it would indicate that the SDK install of VS 2017 is broken somehow.

Tridy, 2017-06-11:
yes, keeping only a reference to 15 in  \Support\frameworkbase\dotNetMF.proj allowed it to build.

I will dig further into 4.3 vs 4.4 in 2017 for Netduino this week. 

Thanks a lot for your help!

valoni, 2017-07-26:
did anyone manage and created VISX for VS2017
 for use of .NET MF 4.3.x /4.4.x

KiwiBryn, 2017-07-26:
Hi

Just about got it working (see attached picture) need to build VM with VS2K17 for testing

Bryn

Tridy, 2017-07-26:
I switched to TinyCLR from GHI Elecronics. I think they are doing a great job at unifying many devices under one OS. I tried it and it seems to works great in VS 2017 with my Netduino 3 WiFi.

https://www.ghielectronics.com/news#24580

https://github.com/ghi-electronics/Docs

doingnz, 2017-07-27:
@Tridy, the question is about getting VS2017 support working for NetMF, an open source project. Proposing a closed source commercial alternative is not going to get the job done.

valoni, 2017-07-27:
Why not having both in VS2017 (Net MF 4.4 and TinyCLROS)

Instead to install two visual studio (2015 and 2017) for development for both frameworks

To use

networkfusion, 2017-07-27:
As I have previously said, nanoframework is worth a look. It has vs2017 support and is completely open source. The more people contribute the quicker it will become mature.

valoni, 2017-07-27:
but there no document where and how to use it 
- what board
- how to install bootloader
- how to install interpreter
- how to use (with VS2017 compatibility with NetFramework ose Core .NET)

Tridy, 2017-07-27:
@doingnz TinyCLR is opensource hosted on GitHub https://github.com/ghi-electronics

doingnz, 2017-07-27:
@Tridy I don't see the source for the GHI prebuilt library i.e. the bulk of .NetMF. There is only a prebuilt library for CortexM4 which may be OK if that is your CPU type. I am using ARM9 HW and there are no sources for the library. By design TinyCLR is understandably for GHI customers.

Tridy, 2017-07-27:
@doingnz You are right! Sorry. My bad. TinyCLR is not open source.

> TinyCLR OS can be ported to new devices. We provide the **precompiled .NET interpreter and runtime** for several architectures. Currently only Cortex M4 is supported, but more will be supported in the future. 

source: http://docs.ghielectronics.com/tinyclr/porting/intro.html

josesimoes, 2017-07-27:
@valoni quick answers to your questions above. For more details please join nanoFramework Slack or open an issue over there. 😉 

> what board
These are the boards currently in the repo (provided as reference): STM32F4_DISCOVERY, STM32F429I_DISCOVERY, STM32F769I_DISCOVERY, NUCLEO64_F091RC, NUCLEO144_F746ZG and Quail. Others can easily be added by the community on a specific repo for those.

> how to install bootloader
Just grab the hex ou bin from the build folder and flash the board.

> how to install interpreter
Just grab the hex ou bin from the build folder and flash the board.

> how to use (with VS2017 compatibility with NetFramework ose Core .NET)
There is an extension for VS2017 (that's all you need to install to develop C# projects). Start a C# nanoFramework project and code as usual. We are trying to follow as close as possible the UWP API.

valoni, 2017-07-29:
@josesimoes 

i have board NUCLEO STM32F411RET6 but where to grab / or create HEX i can not find it anywhere 

i searched and find community contribution done for this board
https://github.com/nanoframework/nf-Community-Targets/tree/19d7909577ae86d4c294584b005c52755835e0ab

but there no instruction how to compile it ????

or ready batch just for execute or complete documentation how to setup for compile step by step 
for all needed files and tools .

i using VS2017 - installed exension 
but no firmware for Nucleo STM32F411RET or could someone provide exact documentation how to build firmware (bootloader) or ready maded bootloader to upload in my board

KiwiBryn, 2017-09-09:
@Tridy @miloush I have been building VSIX with VS2K17 (but not building SDK) using my local NetMF 4.3 install. I have repro with updated templates, a new coredebugvs15, NetMFVS15 (copied VS14 to VS15 folder then updated coredebugvs14, NetMFVS14 etc. as required). 

My VSIX finds my two editions of VS2K17 (two VMs with Community & Enterprise) installed
[image] so my extension manifest changes heading in right direction.

I can create new projects
[image]

But like @Tridy the references are broken and the project won't compile
[image]

[image]

The registry points to 
Computer\HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\.NETMicroFramework\v4.3\AssemblyFolder

The NetMF 4.3 binaries are in 
C:\Program Files (x86)\Microsoft .NET Micro Framework\v4.3Assemblies\
 &
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\Microsoft.Net Micro Framework

I wonder if the path is the problem?

Does anyone have any suggestions as I'm running out of ideas? 

Thanks
Bryn

valoni, 2017-09-09:
Seems it check on     C:\Windows\assembly\  instead required path

KiwiBryn, 2017-09-10:
@valoni the "Add reference" dialog finds the NetMF V4.3 files 

[image]

But adding an assembly like Microsoft.Spot.Native fails with 
[image]

Trying to add mscorlib fails silently. (It's not in the new projects)

I have tried to get VS2K17 support (for V4.3 so I could use my Netduinos) going with multiple versions of the code (I even went back to the CodePlex zip file with the client_v4_3_SDKR2 code). 

Does anyone with more experience with the VSSDK etc. have any suggestions? 

I'll go through the repo tonight looking for anything obvious.

@KiwiBryn

valoni, 2017-09-10:
Can you share client_v4_3_SDKR2 code because i can't download it

KiwiBryn, 2017-09-10:
The [zip ](https://devmobile-my.sharepoint.com/personal/bryn_lewis_devmobile_co_nz/_layouts/15/guestaccess.aspx?docid=080b5ae471a55429293c98da62a6e871e&authkey=AcC3bu-NNBAH2HtHOeWcL7c )file from Codeplex contains a couple of the previous versions as well. 

After some experimentation I found that the number of changes required due to the older versions of Visual Studio supported by this codebase made the VS2K17 VSIX port more complex.

@KiwiBryn
blog.devmobile.co.nz

valoni, 2017-09-10:
@KiwiBryn  thanks a lot for sharing i got now it
and from codeplex i could not adownload it (gave me an gateway message error. ...)

so thanx thanx thanx a lot

[5 more replies]